kennethjiang/OctoPrint-Slicer #126: Use GCODE scripts from octoprint settings

ottojo, 2017-07-08:
The Start GCode script in the advanced settings in the slicer plugin should use the script from the "GCODE Scripts" menu in the octoprint settings.

kennethjiang, 2017-07-08:
Similar to #127, start g-code is taken from selected slicing profile. let me know if this is not the case for you, or you think it's a better idea to take from octoprint settings as opposed to slicing profiles.

stidf, 2017-07-08:
For certain things like extruding temp you would want it to pull from the
slicer profile. However I would pull the printer startup g code from
octoprint rather than the slicer. The desired purge line and printer start
up behavior are generally unique to the printer and some change print to
print. This is why the octoprint slicer has these commands coded into the
printer settings. Basically if you make it slicer dependent the operator
needs to remember to add the start up behavior code to each slicer profile
they make. This is annoying if you have a bunch of slicer profiles as you
have to replicate the code each time. Plus if you make changes to your warm
up routine down the road, you now need to alter every slicer profile,
rather than making changes to the printer printer starting routine settings.

kennethjiang, 2017-07-09:
Now I see the point that @ottojo made. Thanks @stidf for your explanation. I guess because I only have 1 printer, I have never understood why OctoPrint has its own gcode script settings.

The tricky thing is, I don't think it's a good idea to just pull start/end gcode from OctoPrint settings and silently ignore the ones in slicing profile, simply because there are other users (like me) who assume that start/end gcode in slicing profiles will be honored by default. I can see how users are frustrated and confused when they realize it's not the case only after they kick off the print job.

Any idea on how to make this feature intuitive to all users?

stidf, 2017-07-09:
It's not just for multiple printers, think multi materials. As for the
solution, idk because adding the start up routine code to octoprint was one
of the setup steps for getting the onboard slicer working. Either way it's
something that needs to be set on machine profile setup. Generally once
with a little fiddling over the life of the machine. Couldn't you read the
slicer profile on import and ask the user to confirm what it thinks the
start up routine code is?

kennethjiang, 2017-07-09:
@stidf can you help me understand how OctoPrint gcode scripts help with multiple materials? For instance, if you have different start gcode snippets for PLA vs PETG, how do you expect to use OctoPrint gcode scripts to accomplish that?

kennethjiang, 2017-09-03:
Closing due to inactivity.

closed 2017-09-03 by kennethjiang