kennethjiang/OctoPrint-Slicer #127: Use OctoPrint temperature presets

ottojo, 2017-07-08:
Being able to use a temperature preset from OctoPrint instead of manually setting the print temperature on every print would be useful.

kennethjiang, 2017-07-08:
Currently the default temperatures are taken from selected slicing profile. Is this not the case for you?

chronoglass, 2017-09-01:
I'd like to second this request, the slicing profiles are useful for fine tuning the specifics of a material, but being able to select the material from the octoprint presets in the basic tab would allow a little ease of use

John-Mc, 2017-09-01:
Personally, I'd rather have the temps come from the slicing profile, and have the ability to override it (As we do now). We use this in a school setting. So far, 100% of the time we want the kids to use the temperature in the slicing profile. It can be overridden if necessary, but I'd rather have it **default** to the temperatures we have established in the profile that work with that filament.  I prefer that method for my personal printing as well.

Using the profiles for the default temperatures is also consistent with how it works when we are direct connected to the printer and not running OctoPrint. I like the consistency from one method to another.

kennethjiang, 2017-09-02:
So what you want is to have an easy way to pick one of the preset temperatures to overwrite the ones in slicing profile, so that you don't have to manually enter them. Is my understanding correct?

Usually I have different slicing profiles for different materials, to account for the differences not only in temperatures but also in a lot of other settings. Can you elaborate the situations when using different profiles for different materials is not a reasonable solution for you? @chronoglass  @ottojo

chronoglass, 2017-09-16:
I think the individual profiles is a decent fix for the issue, going to my feature request for saving adjustments as a new profile. Could be nice to tie into the "default" material temps of octoprint, but could also become confusing.

kennethjiang, 2018-01-19:
Closing this issue since it seems like a consensus that using slicing profiles is the reasonable way to address it.

closed 2018-01-19 by kennethjiang