kennethjiang/OctoPrint-Slicer #131: Profiles missing when running 2 instances of octopi

kittenkong, 2017-09-02:
I've had the slicer running great with one instance of octopi for a while, using both cura and slic3r engines but after I started running a second instance of octopi the slicer no longer shows any slicing engines or profiles in the drop downs in the new instance or the original one.

kennethjiang, 2017-09-03:
What do you mean by "a second instance"? Did you mean 2 OctoPrint processes running on the same Raspberry Pi, or 2 Rasbperry Pi with 1 Octoprint running on each of them?

kittenkong, 2017-09-03:
That's 2 Octopi processes running on a single Raspberry Pi

kennethjiang, 2017-09-03:
Got it. Can you elaborate on how you set it up so that I can try to reproduce it? I may be able to figure out how to do it myself, but if my way of setting up 2 OctoPrint processes are different than yours, I may not see the same problem as you do.

kittenkong, 2017-09-10:
I'm running a raspbian sdcard (not octopi image), I made copies of the .octoprint file as below
`cd ~
cp -rf .octoprint .octoprint1
cp -rf .octoprint .octoprint2`

Then I made a bash file to start up all my instances at the same time

`nano ~/octoprint_startup.sh
 
#!/bin/sh
 
OCTOPRINT_HOME=/home/username/OctoPrint
 
# start the webui 1
$OCTOPRINT_HOME/run --daemon start
 
# start the webui 2
$OCTOPRINT_HOME/run --daemon start --port 5001 --pid /tmp/octoprint1 --basedir ~/.octoprint1`

kennethjiang, 2017-09-17:
I finally got a chance to replicate your setup. However Slicer plugin worked fine on both OctoPrint instances...

To help me further investigate the issue, can you:

1. Try to load an STL into Slicer plugin.
2. Open Javascript console for the page on which you try to load STL, copy all the messages on Javascript console and attach them to this issue? In case you are not familiar with it, this is how you can do it in Chrome: https://developers.google.com/web/tools/chrome-devtools/console/
3. Login to your Pi, and find file `octoprint.log` (located in `basedir/logs`) in both of your OctoPrint installation. Attach them here.

Thanks @kittenkong !

kittenkong, 2017-09-18:
No fair, I tried making instances with on octopi image and again installing octoprint from scratch on a raspbian install and have had the same problem.
[kittenkonglogs.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/1310596/kittenkonglogs.zip)

Here are the logs, I couldn't actually load an STL as all the button are greyed out.

kittenkong, 2017-09-18:
...

I am a dumb dummy. After correctly linking to curaengine the "slice" button activates and everything is good. I am not sure why I had the problem the first time as i had everything working and slicing properly until I ran a second instance and it all failed. It definitely works with four instances now though.

kennethjiang, 2017-09-18:
Glad to know you got it figured out. @kittenkong 

I made the similar mistake and I should have warned you for that...

closed 2017-09-18 by kennethjiang