kennethjiang/OctoPrint-Slicer #132: Slicing failed

rtsense, 2017-09-06:
with octoprint and cura profile I have this error
Slicing failed Could not slice BI.tmp.1504687229640.stl to BI.gco: Unknown error, please consult the log file
where the log file is located ?

kennethjiang, 2017-09-06:
Hi @rtsense - you can move your model on the build plate just a little bit and slice again. For some unknown reason Cura some times fails to slice a model when it's placed at the origin (0,0).

If it still fails, please attached the STL and I will try it on my end.

rtsense, 2017-09-08:
Hi @kennethjiang - I tested your solution and several stl that did not solve the problem. I installed the plugin on another printer and this time it works.
The profile cura not being the same may be that it comes from there

kennethjiang, 2017-09-08:
Have you ever been able to slice an STL on your 1st printer? If not, it's very likely that the Cura installed on that printer is somehow corrupted.

The best way to find out what happened is to look at the log files. If you know how to SSH to the Raspberry Pi of your 1st printer, you can find the log files in folder `$HOME/.octoprint/logs`. In that folder you will find files with name like `octoprint.log`, or `octoprintxxx.log`. If possible, please zip all these files and send them to me at kenneth dot jiang at gmail dot com.

rtsense, 2017-09-08:
If I disable your plugin I can slicer without problem

rtsense, 2017-09-08:
log after slicing error : 

2017-09-08 18:38:02,753 - octoprint.plugins.cura - ERROR - Could not slice via Cura, got an unknown error
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/cura/__init__.py", line 306, in do_slice
    p = sarge.run(args, cwd=working_dir, async=True, stdout=sarge.Capture(), stderr=sarge.Capture())
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg/sarge/__init__.py", line 1407, in run
    p = Pipeline(cmd, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg/sarge/__init__.py", line 975, in __init__
    self.source = ' '.join(source)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 253: ordinal not in range(128)

kennethjiang, 2017-09-09:
From the error my best guess is that the slicing profile has some special character that is causing the problem.

To help me dig deeper, please:

- Attach the slicing profile that you were using to slice the model;
- Open OctoPrint in Chrome. Activate Developer Tools in Chrome. Enable Slicer plugin to slice an object. Switch to `Network` in Developer Tools. Find an entry similar to the one highlighted in the screenshots. Right-click that entry and select "Copy as cURL". Paste the content to this issue.
- Post the logs immediately before this error as well?

[image]
[image]

kennethjiang, 2017-09-09:
I'll be happy to do a google hangout session to walk you through all these steps if that's easier for you.

rtsense, 2017-09-09:
[conf-01-09-2017.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/1289603/conf-01-09-2017.zip)

rtsense, 2017-09-09:
curl "http://192.168.0.17/api/files/local/The_Ultimate_Spool_Holder_v2.tmp.1504941967425.stl" -H "Cookie: mp_ade6b64c4db77172707571c1f9cfcebe_mixpanel=^%^7B^%^22distinct_id^%^22^%^3A^%^20^%^2215e568e51dd5f6-0eb15f8ca92b26-1781c36-232800-15e568e51de5d5^%^22^%^2C^%^22^%^24initial_referrer^%^22^%^3A^%^20^%^22^%^24direct^%^22^%^2C^%^22^%^24initial_referring_domain^%^22^%^3A^%^20^%^22^%^24direct^%^22^%^7D; mp_mixpanel__c=0; session_P80=.eJx9jkELgjAYQP9KfGeJiLoMvIVCkDK0xnYZppNNp4XbkE3879XNU-fHe7wFeDsJIwHZyYkIuGoALbB7AoIs0FNG6JmVtWcDPeTkFtjl2tMj9jTcPe2SgXVS5ymOYY1ANWK0yvp95azk1r8FoNFpvSGbOksfoSKN-5nOiIn_YUYYo17jVq9LiUk_z0WauALH34H1A2FkRDE.DJUpDw.CGTGqugVvo4cevOBGCZ2z6lXYJk" -H "Origin: http://192.168.0.17" -H "Accept-Encoding: gzip, deflate" -H "Accept-Language: fr-FR,fr;q=0.8" -H "User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/61.0.3163.79 Safari/537.36" -H "Content-Type: application/json; charset=UTF-8" -H "Accept: application/json, text/javascript, */*; q=0.01" -H "Cache-Control: no-cache" -H "X-Requested-With: XMLHttpRequest" -H "Connection: keep-alive" -H "Referer: http://192.168.0.17/" -H "X-Api-Key: B5B517653A5742C28AC930BE3C023F9F" --data-binary "^{^\^"command^\^":^\^"slice^\^",^\^"slicer^\^":^\^"cura^\^",^\^"profile^\^":^\^"conf-01-09-2017^\^",^\^"printerProfile^\^":^\^"_default^\^",^\^"destination^\^":^\^"The_Ultimate_Spool_Holder_v2.gco^\^",^\^"position^\^":^{^\^"x^\^":100.00000381469727,^\^"y^\^":100^},^\^"profile.retraction_enable^\^":true,^\^"profile.fan_enabled^\^":true,^\^"profile.layer_height^\^":0.2,^\^"profile.print_bed_temperature^\^":60,^\^"profile.fill_density^\^":17,^\^"profile.wall_thickness^\^":1.2,^\^"profile.print_speed^\^":40,^\^"profile.solid_layer_thickness^\^":1.2,^\^"profile.travel_speed^\^":80,^\^"profile.outer_shell_speed^\^":40,^\^"profile.inner_shell_speed^\^":0,^\^"profile.infill_speed^\^":0,^\^"profile.bottom_layer_speed^\^":15,^\^"profile.filament_flow^\^":100,^\^"profile.retraction_speed^\^":120,^\^"profile.retraction_amount^\^":3.45,^\^"profile.fan_full_height^\^":0.2,^\^"profile.fan_speed^\^":30,^\^"profile.fan_speed_max^\^":90,^\^"profile.print_temperature^\^":^[210^],^\^"profile.start_gcode^\^":^[^\^"^\^\n;Filament used: ^{filament_amount^}m ^{filament_weight^}g^\^\n;Filament cost: ^{filament_cost^}^\^\n;M190 S^{print_bed_temperature^} ;Uncomment to add your own bed temperature line^\^\n;M109 S^{print_temperature^} ^\^\t; Uncomment to add your own temperature line^\^\nG21        ^\^\t^\^\t^\^\t^\^\t^\^\t; D^éfinir les unit^és en Millim^ètres^\^\nG1 F3000 ^\^\t^\^\t^\^\t; Vitesse de d^éplacement^\^\nG28 X Y^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Home X puis Y^\^\nG91^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Position Relative^\^\nG1 Z1^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Remonter la tete de 20 mm^\^\nG1 X30^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Potitionement au coin supperieur gauche du plateau^\^\nG90 ^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Position Absolue^\^\nG92 X0^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Red^éfinition de l'origine^\^\nG1 X100 Y180^\^\nG28 Z^\^\n;G29^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Detailed Z-Probe ^– palpage du plateau pour cr^éer un plan dynamique^\^\nM109 S^{print_temperature^} ^\^\t; D^éfinir la temp^érature de l'extrudeur et attendre^\^\nM82^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Set extruder to absolute mode^\^\nM107^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Start with the fan off^\^\nG91 ^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Zelative positioning^\^\nG1 Z-10^\^\t^\^\t^\^\t^\^\t^\^\t; OFFSET de reglage hauteur de buse (l^’offset peut ^être positif ou n^égatif); ex: Z0.5 ou Z0.1 ou Z-0.4 ou Z-1.3)^\^\nG90^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Position Absolue^\^\nG92 Z0^\^\nG92 E0 ^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Zero the extruded length^\^\nG1 F200 E3 ^\^\t^\^\t^\^\t^\^\t^\^\t; Extrude 10mm of feed stock^\^\nG92 E0 ^\^\t^\^\t^\^\t^\^\t^\^\t^\^\t; Zero the extruded length again^\^\nG1 Z0^\^\n;M106 ;start fan full power^\^\n;M109 S^{print_temperature^}^\^"^],^\^"profile.end_gcode^\^":^[^\^"^\^\nM104 S0                     ;extruder heater off^\^\nM140 S0                     ;heated bed heater off (if you have it)^\^\nG91                                    ;relative positioning^\^\nG1 E-1 F300                            ;retract the filament a bit before lifting the nozzle, to release some of the pressure^\^\nG1 Z+0.5 E-5 X-20 Y-20 F^{travel_speed^} ;move Z up a bit and retract filament even more^\^\nG28 X0 Y0                              ;move X/Y to min endstops, so the head is out of the way^\^\nM84                         ;steppers off^\^\nG90                         ;absolute positioning^\^\n;^{profile_string^}^\^"^],^\^"profile.filament_diameter^\^":^[1.73^],^\^"profile.support^\^":^\^"none^\^",^\^"profile.platform_adhesion^\^":^\^"none^\^",^\^"_bindings^\^":^[^\^"^#basic_overrides^\^",^\^"^#advanced_overrides^\^"^],^\^"_unbound^\^":false,^\^"profileAjax^\^":^{^\^"readyState^\^":4,^\^"responseText^\^":^\^"^{^\^\^\^"data^\^\^\^": ^{^\^\^\^"bottom_layer_speed^\^\^\^": 15, ^\^\^\^"bottom_thickness^\^\^\^": 0.25, ^\^\^\^"brim_line_count^\^\^\^": 20, ^\^\^\^"cool_head_lift^\^\^\^": false, ^\^\^\^"cool_min_feedrate^\^\^\^": 10, ^\^\^\^"cool_min_layer_time^\^\^\^": 5, ^\^\^\^"end_gcode^\^\^\^": ^[^\^\^\^"^\^\^\^\nM104 S0                     ;extruder heater off^\^\^\^\nM140 S0                     ;heated bed heater off (if you have it)^\^\^\^\nG91                                    ;relative positioni [... cut ...] ,^\^"support_z_distance^\^":0.15,^\^"travel_speed^\^":80,^\^"wall_thickness^\^":1.2,^\^"wipe_tower^\^":false,^\^"wipe_tower_volume^\^":15^},^\^"default^\^":true,^\^"description^\^":^\^"Imported from conf-01-09-2017.ini on 2017-09-01 19:08^\^",^\^"displayName^\^":^\^"conf-01-09-2017^\^",^\^"key^\^":^\^"conf-01-09-2017^\^",^\^"resource^\^":^\^"http://192.168.0.17/api/slicing/cura/profiles/conf-01-09-2017^\^"^},^\^"status^\^":200,^\^"statusText^\^":^\^"OK^\^"^},^\^"booleans^\^":^{^}^}" --compressed

rtsense, 2017-09-09:
2017-09-08 15:55:06,194 - octoprint.server - INFO - --- Log roll over detected ---------------------------------------------------
2017-09-08 15:55:06,195 - octoprint.server - INFO - OctoPrint 1.3.4 (master branch)
2017-09-08 15:55:06,196 - octoprint.plugin.core - INFO - 16 plugin(s) registered with the system:
|  Announcement Plugin (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/announcements
|  Core Wizard (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/corewizard
|  CuraEngine (<= 15.04) (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/cura
|  Custom Background (0.5.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_custombackground
|  Custom Control Editor (0.2.1) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_customControl
|  Cyborg Theme (0.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_cyborgtheme
|  Discovery (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/discovery
|  Fullscreen Plugin (0.0.3) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_fullscreen
|  Gcodebar Plugin (0.1.4) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_GCodeBar
|  Plugin Manager (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/pluginmanager
|  Printer Stats (1.0.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_stats
| !Slic3r (1.1.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_slic3r
|  Slicer (1.3.0) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_slicer
|  Software Update (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/softwareupdate
| !Tempsgraph Plugin (0.2.2) = /home/pi/oprint/local/lib/python2.7/site-packages/octoprint_tempsgraph
|  Virtual Printer (bundled) = /home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.3.4-py2.7.egg/octoprint/plugins/virtual_printer
2017-09-08 15:55:06,197 - octoprint.server - INFO - ------------------------------------------------------------------------------
2017-09-08 15:55:06,191 - octoprint.util.pip - INFO - pip installs to /home/pi/oprint/lib/python2.7/site-packages/, --user flag needed => no, virtual env => yes
2017-09-08 15:55:08,644 - octoprint.server.preemptive_cache - INFO - ... done in 103072.17s
2017-09-08 15:55:08,645 - octoprint.server.preemptive_cache - INFO - Preemptively caching / (ui _default) for {'query_string': 'l10n=fr', 'path': '/', 'base_url': 'http://192.168.0.17/'}
2017-09-08 15:55:12,304 - octoprint.plugins.softwareupdate - INFO - Saved version cache to disk
2017-09-08 15:55:13,174 - octoprint.server.preemptive_cache - INFO - ... done in 4.53s
2017-09-08 15:56:03,505 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.0.11
2017-09-08 15:56:05,905 - octoprint.plugins.announcements - INFO - Loaded channel _important from http://octoprint.org/feeds/important.xml in 0.22s
2017-09-08 15:56:06,476 - octoprint.plugins.announcements - INFO - Loaded channel _releases from http://octoprint.org/feeds/releases.xml in 0.21s
2017-09-08 15:56:06,760 - octoprint.plugins.announcements - INFO - Loaded channel _blog from http://octoprint.org/feeds/octoblog.xml in 0.22s
2017-09-08 15:56:06,986 - octoprint.plugins.announcements - INFO - Loaded channel _plugins from http://plugins.octoprint.org/feed.xml in 0.14s
2017-09-08 15:56:07,316 - octoprint.plugins.announcements - INFO - Loaded channel _octopi from http://octoprint.org/feeds/octopi.xml in 0.24s
2017-09-08 15:56:11,294 - octoprint.filemanager.analysis - INFO - Starting analysis of local:cube.gcode
2017-09-08 15:56:12,023 - octoprint.filemanager.analysis - INFO - Analysis of entry local:cube.gcode finished, needed 0.73s
2017-09-08 15:56:35,683 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-09-08 16:14:39,921 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-09-08 16:29:37,394 - octoprint.util.comm - INFO - Finished in 897.503 s.
2017-09-08 16:38:14,799 - octoprint.filemanager.analysis - INFO - Starting analysis of local:cube.gcode
2017-09-08 16:38:15,606 - octoprint.filemanager.analysis - INFO - Analysis of entry local:cube.gcode finished, needed 0.81s
2017-09-08 16:38:20,948 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-09-08 16:57:19,270 - octoprint.util.comm - INFO - Finished in 1138.340 s.
2017-09-08 17:34:12,585 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-09-08 17:41:06,660 - octoprint.util.comm - INFO - M110 detected, setting current line number to 0
2017-09-08 17:54:22,207 - octoprint.util.comm - INFO - Finished in 795.555 s.
2017-09-08 18:31:53,254 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.0.11
2017-09-08 18:31:53,650 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.0.11
2017-09-08 18:33:08,659 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.0.11
2017-09-08 18:33:08,968 - octoprint.server.util.sockjs - INFO - New connection from client: 192.168.0.11
2017-09-08 18:34:21,181 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.0.11
[... cut ...]
    p = sarge.run(args, cwd=working_dir, async=True, stdout=sarge.Capture(), stderr=sarge.Capture())
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg/sarge/__init__.py", line 1407, in run
    p = Pipeline(cmd, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sarge-0.1.4-py2.7.egg/sarge/__init__.py", line 975, in __init__
    self.source = ' '.join(source)
UnicodeDecodeError: 'ascii' codec can't decode byte 0xc3 in position 253: ordinal not in range(128)

rtsense, 2017-09-09:
[files.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/1289614/files.zip)

kennethjiang, 2017-09-09:
Thank you so much @rtsense will look into it soon!

eyal0, 2017-09-10:
I see a lot of french in the comments in conf-01-09-2017.ini, including non-ASCII characters, like "e" with acute accent.  Try removing all of those non-ascii characters @rtsense  ?

rtsense, 2017-09-11:
Thank you 
I suspected that it came from my profile 
The problem come from commenters in French in the start Gcode
Thanks again to you two
your plugin is truly top

rtsense, 2017-09-11:
On Cyborg Theme your rotate png has some white pixel that degrade the aspect
you can use this one if you want 
[image]

kennethjiang, 2017-09-11:
Thank you @rtsense for providing the details! I'm glad that you get this figured out.

Really appreciated your help @eyal0 while I couldn't spend time on this issue!

kennethjiang, 2017-09-11:
Also appreciated the icon you design for me @rtsense It worked pretty well in dark theme but is not 100% consistent with normal theme. I will check with the original designer to see if he can fix the pixelation for me. If not I'll be happy to adopt your version. :)

kennethjiang, 2017-09-18:
Hi @rtsense I just released version 1.3.1 that includes icons compatible with Cyborg theme. Please upgrade it. Happy slicing. :)

closed 2017-09-18 by kennethjiang