kennethjiang/OctoPrint-Slicer #133: "Split into part" failed for particular STL

kennethjiang, 2017-09-07:
The attached STL is cut in 2 halves by 3D builder. Slic3r is able to recognize the cut and split them into 2 parts. However "split into part" command in Slicer plugin fails to do so.

Cura failed the test too. 
[lungo.stl.gz](https://github.com/kennethjiang/OctoPrint-Slicer/files/1285989/lungo.stl.gz)

eyal0, 2017-09-11:
This is a really hard problem.  I have worked on it before.  The current way that 3tk isolates is by looking for common intersections.  That only works if there is space between the models.  But if you use a problem is cut a model, it won't leave space.

The solution for that is to look at edges instead.  Two faces are connected only if they share a common edge and that edge has to be in the reverse direction.  For example:

[image]

Because of the right-hand rule for STL files, we see that for both triangles have normals pointing out of the screen.  And we can also see that the common edge is c-b for the blue triangle, which is backward in the alphabet, and for the red triangle, it's a-b, which is forward in the alphabet.  So that means that the edges are connected.  If the blue triangle were cba instead of abc then the edges would have reverse direction and it would be a bad connection.

[image]

Here there are two triangles connected to each other.  The blue common edge can't be connected to the outside red edges because the normals aren't pointing in a consistent direction.  It's like the outside is connected to the inside and that's not legal in STL.  So it's possible to split this STL apart.

But what about this case?

[image]

On the left is all the lines and their normal vectors.  There are two ways to split it, both legal.  One is two pentagons.  The second is a diamond instead a 6-sided shape.  Both follow all the rules.

So there's no general method to split shapes.

The STL file that you provided is extra difficult because it also has degenerate triangles with area 0!  I wrote an algorithm that does splitting with reverse connected edges but it doesn't work on your STL file.  My attempt is here: https://github.com/eyal0/3tk/tree/split_shapes_by_edge

eyal0, 2017-09-11:
This is an example of a file with two shapes that I'm able to split using the branch that I mentioned above.  It's two pyramids with triangular bases connected by one face.

[triangletest.stl.gz](https://github.com/kennethjiang/OctoPrint-Slicer/files/1293782/triangletest.stl.gz)

kennethjiang, 2017-09-11:
Thank you so much @eyal0 for the detailed explanation and the work you have done! I will play with your branch when I get a chance.

eyal0, 2017-09-16:
I did some more work and now it's able to slice the lungo file, too.  I'll explain how it works.

First, we find all possible neighbors of all faces.  For example, these two triangles share an edge:

[image]

But the faces can only be connected if they have the same normal.  So either you're looking at the outside of the shape for both triangles or the inside for both.  If one of those is outside of the shape and one inside, it can't be connected.  For each face, for each edge, we create a set of all the possible connections.

Next, we use our first heuristic: **make all forced connections**.  In this diagram, the edges are colored and the outward-facing side on each edge is marked with a black dot:

[image]

Where the arrows are, there is only one options for connection.  We connect those first because they have to be connected.  We do all forced connections until there are none left.

Now the question is what to do about the other edges.  There are two ways that we could connect it:

Red to green and orange to blue:
[image]

Or blue to green and red to orange:
[image]

We need a heuristic to choose the better choice because both are legal (shared edge and normals match).  The heuristic here is **choose the connection that doesn't join shapes**.  In this case, we already have 2 partial shapes: the one on the left and the one on the right.  If we connect the first way, they will join into a single shape.  But our goal is to split, so we choose the second way to make two triangles.

The way that we calculate it is we find all the unconnected edges per shape.  I call this the frontier.  The frontier on the left triangle is one side of the green and one side of the blue.  Because the green and blue are already connected to the same partial shape, we choose to connect those.  After we connect those, we go back to the plan of connecting forced edges.  Each edge can only be used once, so because blue is already used, orange can only connect to red.

We keep doing that until we get a case where we can't do either of the above and then we use the third heuristic: **Don't connect slivers.**  Slivers are when we have two faces that are very, very close to each other, making a shape with no volume.  For example:

[image]

We only have two forced connections, marked by arrows.  We There are two ways to connect this:

Two triangles that share an edge:
[image]

One diamond and one sliver:
[image]

We can't use the second heuristic because the frontier of the left shape, yellow and blue, can't be connected directly.  So we look for slivers.  We loop through all the possible face connections to see if there are any that are very close to 0 degrees or to 360 degrees.  (We look for the connection that is farthest from 180 degrees.)  In this case, we see that connecting purple and green connects two faces that are 360 degrees apart.  That would make the thin grey shape in the second image.  That's probably the wrong thing to do so we remove that possible connection and start over.  With that one removed, we start over and look for forced connections and then frontier connections.  The final algorithm looks like this:

```
while unnconectedEdges.size > 0:
  make all forced connections
  if at least one forced connection was made:
    continue
  make all frontier connections
  if at least one frontier connection was made:
    continue
  find the connection with the angle that is farthest from 180 degrees (closest to 0 or 360)
  remove that connection
```

If the input STL is legal, this algorithm should always eventually end.  In a single object that can't be split, the edge of each face matches the edge of another face.  If multiple objects are put in the same file then everywhere that they have edges in common they are always in pairs with the matching edge.  So no matter what order you connect the edges, you should always have a matching edge to use.  The heuristics above are to resolve connections so that the shapes created make the most sense.

I wrote the code and added tests into the test directory, including lungo, the triangles from above, and also a really big file for testing speed.  The very large file takes about 17 seconds to split.  That's about double of how long the old algorithm takes for splitting.  Much of that is spent calling keyForTrio.  I'll try to make keyForTrio faster.

closed 2017-09-17 by kennethjiang