kennethjiang/OctoPrint-Slicer #134: slic3r plugin is not detected or offered

cnlson, 2017-09-08:
I have installed both slic3r 1.2.9 and your slicer viewer. Cura, however,  is the only clicer detected

with an stl loaded
[image]

settings plugins - slic3r plugin present
[image]

slic3r plugin setup correctly
[image]

devRhumbix, 2017-09-08:
Thank you @cnlson for reporting this issue.

Assuming you have done things like rebooting OctoPrint and force-refreshing page, please do the following steps to help me investigate:
1. Open OctoPrint in Chrome. Open developer tools, and look for Javascript errors on console. Post errors here if you see any.
2. Refresh the page and look for the network call as highlighted in the attached screenshot. Copy the content in the highlighted box and post it here.

Thanks!

[image]

cnlson, 2017-09-08:
I ended up making it work, sorry for not closing the issue. in the slic3r
plugin, after you have slic3r installed and after you upload a profile you
need to go back and enable the profile. as soon as i did that and rebooted
it worked.

thanks again! Really enjoy the slicer in octoprint. I can slice from my
mobile now

<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=icon>
Virus-free.
www.avast.com
<https://www.avast.com/sig-email?utm_medium=email&utm_source=link&utm_campaign=sig-email&utm_content=webmail&utm_term=link>
<#DAB4FAD8-2DD7-40BB-A1B8-4E2AA1F9FDF2>

closed 2017-09-08 by kennethjiang