kennethjiang/OctoPrint-Slicer #136: Show raft, brim and supports in preview

leobel96, 2017-10-02:
Hello, Is it possible to add something similar to the "layers view" present in Cura so that Can be seen rafts, brims, supports,... directly in the preview such as shown in the picture below? Thanks
[image]

kennethjiang, 2017-10-02:
Hi @leobel96 thank you for proposing this idea.

Show rafts/brims etc is part of g-code viewing. Cura and some other slicer do a pretty good job to provide a seamless transition between STL viewing and g-code viewing. However doing the same in Slicer plugin will require a significant amount of work.

My workaround is to slice STL into gcode. And then load gcode into gcode-viewer. Will this workaround be ok for you too?

leobel96, 2017-10-02:
What about only showing rafts, brims and supports without analyze g-code but only adding it in the preview using a sort of  "prevision" of where they should appear?

kennethjiang, 2017-10-02:
Not sure what you mean by "prevision". But I assume the reason why you want to see raft/brim/skirt is to make sure they are what you expected. 

How I think raft/brim/skirt *should* appear may not be how Cura/Slic3r *actually* generates it. And therefore IMO showing it will be misleading to say the best. Plus it won't easy to code it up.

 Did I overlook some easy way to reverse-engine exactly how Cura and/or Slic3r would generate raft/brim/skirt?

leobel96, 2017-10-02:
Everything is exact, you are right. I'll use G-CODE viewer, no problem.

closed 2017-10-02 by leobel96