kennethjiang/OctoPrint-Slicer #138: No brim generated

leobel96, 2017-10-12:
Hello, I select "Adhesion : brim" usind the dropdown menu but no brim is generated during print. How is it possible?

kennethjiang, 2017-10-12:
Did you use Cura or Slic3r? Can you upload the STL and gcode so that I can take a look?

leobel96, 2017-10-12:
I used Cura. These are the files 
[files.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/1380125/files.zip)

kennethjiang, 2017-10-12:
The brim was actually generated in your gcode. Attached is the screenshot of the 1st and 2nd layer of your gcode. 

The brim in your gcode is a lot thinner than the one I generated using your STL. I suspect it's because you customized "brim line amount" setting in the profile.

[image]
[image]

leobel96, 2017-10-12:
Yes, I'm sorry, my fault. Have a nice day!

closed 2017-10-12 by kennethjiang