kennethjiang/OctoPrint-Slicer #141: Support for Selecting Tool for multiple STL Prints

mriveralee, 2017-10-29:
I have a 2 extruder setup and frequently print things with mixed-materials. It would be great if there could be support for selecting the Tool for a particular STL from within the viewport.

kennethjiang, 2017-10-31:
Currently Slicer plugin is using Cura 15.04 running inside OctoPrint as slicing engine. I don't think Cura 15.04 supports dual extruders. But let me know if I'm wrong @mriveralee

mriveralee, 2017-11-02:
I did some digging: 

Cura 15.04 also supports dual extrusion by a command in the interface called "Dual Extrusion Merge". After importing two models, you can select one, right click, and select "Dual Extrusion Merge". Cura then labels the initially selected one to be printed with "Extruder 1". 

There's a post on the forums describing how to do this via the older [Cura software](https://ultimaker.com/en/community/10133-how-to-design-part-for-two-extruders) - (a quick search on the page for "Dual Extrusion Merge" will highlight the post describing the process with photos. I downloaded the Cura software (v15.04) and indeed I was able to do this. 

I'm not sure how this translate into how OctoPrint sends commands, but it would seem it's definitely possible to do.

kennethjiang, 2017-11-02:
Great thank you @mriveralee  for doing the digging for me. I will take a look at if the same is doable in OctoPrint.

kennethjiang, 2017-12-01:
Sorry @mriveralee for the delay. I was just back from an extended vacation a few days ago. Am trying to catch up on everything.

I suspect that *Dual Extrusion Merge* function is not exposed to Cura's command line executable. When I did `cura_engine -h`, I got:
*usage: CuraEngine [-h] [-v] [-m 3x3matrix] [-c <config file>] [-s <settingkey>=<value>] -o <output.gcode> <model.stl>*

It doesn't seem to take any argument that allows it to accomplish dual extrusion.

It'll be great if you can help me dig a bit more to find out if I missed anything. Meanwhile I will do more research on my side. Thanks a lot for your help @mriveralee !

kennethjiang, 2017-12-05:
Hi @mriveralee: dual extrusion merge doesn't seem to be available in Cura 15.04 for Mac either. Please see attached screenshot.
[image]

I suspect this is a feature available only in the propriatary part of Cura. Since what's running inside OctoPrint is a Cura command line, I don't think I can incorporate this feature, unfortunately. Let me know if you find otherwise.

mriveralee, 2017-12-05:
Thanks for digging into this, I'll see if I can find any more information or another way to do this using the Command line.

kennethjiang, 2017-12-05:
Perfect thanks @mriveralee !