kennethjiang/OctoPrint-Slicer #142: Scaling problem

thjansen72, 2017-11-05:
I loaded a model into the slicer. No problem. But I need it a little smaller. scaling  - no matter if single axis or all 3 locked - no problem.
But: if i rotate the model  befor scaling  (it is vrtical in the project, so I rotate it so it can lay flat)  and then try to reduce i.e. the z-hight (just a prototyp - don't need all the material to be printed, only the perimeters), nothing is o.k.
Scaling all 3  dimensions (lock) is no problem.  Scaling a single axis is a little goofy...
It seems the rotating keeps the dimensions orientation, but the scaling don't really mind and is scaling the orientation, wich it "thinks", it might be the best... but not  the original  or the workspace orienation.
It seems to get a bit dizzy with the rotation befor scaling .
[image]
[image]
[image]

Now i know, it's no problem to do the scaling first, then the rotation. But if you don't know...

kennethjiang, 2017-11-15:
Sorry @thjansen72 for the delay. I have been very tied up with another project.

I tried to scale Z axis and then rotate the part and I didn't see any problem. Please see the attached screenshot. Did I misunderstood the problem you described?

[image]
[image]

thjansen72, 2017-11-15:
[image]
Hmm I had to get clear about what really happens.
Lets say the orientation of the bed is  x->front-line (to the sides), y ->left+right line (to the back), z- up).
When i place a model, it "gets" the same coordinate-system of the bed.
After rotating the model, its coordinate-system is also rotating.
In your example you changed the value of the x-axis, wich was not rotated - try Y or Z....
In the scaling-dialog I'd think all orientations are like the bed-orientation, but by rotating  the coordiante system of the modell  (because of rotating it)  (in this example)  x  - is like before, y and z switched.
Entering a new value for y or z  will chause in the change of the lenght-value-field of the other orientation.
Irritating...  (I need a dictonary and a thesaurus "english" for chrome ....)
[image]
[image]

kennethjiang, 2017-11-21:
Thank you for the detailed info @thjansen72 ! I'm currently on vacation with very limited bandwidth. Will take a look at it next week when I'm back.

kennethjiang, 2017-12-01:
Sorry for the delay @thjansen72 . I was just back from vacation.

I'm not sure I understand what the problem is. Can you point out what you think is wrong in the screenshots you poste above?

I also did some other tests and the screenshots are as follows:

*0. Original part placement and 1.0 scale:*
[image]

*1.1 First rotate 90 degrees on both Y and Z:*
[image]

* 1.2 Then scale Z by 2x:*
[image]

*2.1 First scale Z by 2x:*
[image]

* 2.2 Then rotate 90 degrees on both Y and Z:*
[image]

All the steps seem to behave as I expected.

kennethjiang, 2017-12-12:
@thjansen72 can you help me understand what the problem is? I am pretty sure I misunderstood you but I can't figure out where the misunderstanding occurred.

John-Mc, 2017-12-12:
Not sure if I'm following what @thjansen72 is saying, but let me take a stab at it.

In the example @kennethjiang  gives in his post with the sort of triangle shaped thing: in picture 1.1 you rotate 90˚ around both Y and Z. But then when you scale Z, it scales the **original** Z-axis, but @thjansen72 is expecting it to scale along the presently displayed Z-Axis, so that the shortest dimension (which is the _current_ Z) would be made larger.

When all the rotations are 90˚, it's easy to adjust, but when you are rotating by some odd number, it gets tougher.

Ken - I think we had a similar discussion some time ago, but I can't remember if it was scaling or lay flat or something else (it may have been issuing a lay flat after rotating doing funny things?) I'd get something positioned, almost how I wanted it, then issue a command (lay flat?) and it would undo some of the rotations I had done.

kennethjiang, 2017-12-12:
Thank you @John-Mc for helping us out here. I think your interpretation makes sense. As a matter of fact I just tested it in Cura and it worked exactly as what you described @John-Mc .

I will do some research to see how difficult it is to change the behavior to be consistent with Cura. 

I think the previous discussion we had was #81. It is a different issue and I know it's much harder to do.

kennethjiang, 2018-01-19:
Closing issue b/c of inactivity.

closed 2018-01-19 by kennethjiang