kennethjiang/OctoPrint-Slicer #143: display the slicing percentage in the progress bar

mrouillard, 2017-11-11:
It would be nice to have the percentage of slicing displayed in the status bar, similarly like done with the default Cura slicer.
I don't know if this is feasible, as I suppose it depends on the capabilities of the Slic3r program. But it would give a better user experience when slicing big models.

kennethjiang, 2017-11-15:
My Slicer plugin is only a front-end for Cura and Slic3r engines running inside Paspberry Pi. In particular, Slic3r engine is implemented in another project here: https://github.com/OctoPrint/OctoPrint-Slic3r

I think you need to open an issue with that project to request the progress bar feature. Let me know if it doesn't make sense to you @mrouillard

mrouillard, 2017-11-15:
Ok, this makes sense. Thanks!

closed 2017-11-15 by mrouillard