kennethjiang/OctoPrint-Slicer #144: Never seems to slice with slic3r plugin

gappleto97, 2017-11-15:
Latest version of all components. No log events seem to be issued.

kennethjiang, 2017-11-15:
Can you give me a bit more info about this bug @gappleto97 ? 

Did you install the Slic3r plugin? It is available here: https://github.com/OctoPrint/OctoPrint-Slic3r

Were you able to see `Slic3r` as an option when you use Slicer Plugin?

gappleto97, 2017-11-15:
I was able to see it as an option, but when I hit slice no processes were started on the computer, and no log events were emitted.

I installed both via the built in plugin manager.

kennethjiang, 2017-11-15:
If you uninstall Slicer Plugin, will you be able to use Slic3r successfully?

gappleto97, 2017-11-15:
Yes

kennethjiang, 2017-12-01:
Sorry for the delay @gappleto97 . I just came back from an extended vacation.

I suspect that the problem you are experiencing is caused by an error in browser. Can you please open the OctoPrint page in Chrome, load an STL in Slicer plugin? Then [open Chrome Javascript console](https://stackoverflow.com/questions/66420/how-do-you-launch-the-javascript-debugger-in-google-chrome), copy the content in Javascript console and post it in the comment. Thanks!

gappleto97, 2017-12-02:
Would it make a difference if I ran it in Firefox? Its a mild pain to install chrome on Linux.

kennethjiang, 2017-12-02:
Yes Firefox should work too @gappleto97 . What I need is the Javascript console output.

kennethjiang, 2017-12-04:
Were you able to capture the Javascript console output @gappleto97 ?

kennethjiang, 2017-12-12:
Do you still have the problem @gappleto97 ?

kennethjiang, 2018-01-19:
Closing issue b/c of inactivity.

mikedmor, 2018-01-27:
Also having this issue. Found this error in the console on chrome

`Uncaught TypeError: Cannot read property 'replace' of undefined
    at packed_plugins.js:36459
    at Pn (packed_libs.js:10658)
    at Function.<anonymous> (packed_libs.js:10679)
    at OverridesViewModel.self.updateOverridesFromProfile (packed_plugins.js:36458)
    at Object.success (packed_plugins.js:36513)
    at fire (packed_libs.js:3188)
    at Object.fireWith [as resolveWith] (packed_libs.js:3318)
    at done (packed_libs.js:8758)
    at XMLHttpRequest.<anonymous> (packed_libs.js:9124)
    at XMLHttpRequest.d (raven.min.js:2)
raven.min.js:2 Uncaught TypeError: result$$1[key].replace is not a function
    at packed_plugins.js:36557
    at Pn (packed_libs.js:10658)
    at Function.<anonymous> (packed_libs.js:10679)
    at OverridesViewModel.self.toJS (packed_plugins.js:36556)
    at SlicerViewModel.self.sliceRequestData (packed_plugins.js:37218)
    at SlicerViewModel.self.slice (packed_plugins.js:37258)
    at HTMLButtonElement.<anonymous> (packed_libs.js:10845)
    at HTMLButtonElement.dispatch (packed_libs.js:4738)
    at HTMLButtonElement.elemData.handle (packed_libs.js:4550)
    at HTMLButtonElement.d (raven.min.js:2)`

mikedmor, 2018-01-27:
Wanted to chime in that i was able to get this working. Not sure if it was missed in an update or something important, but i was able to get it to work by commenting out the following lines from ~/.octoprint/generated/webassets/packed_plugins.js

line 36457-36461
`                if (self.slicingViewModel.slicer() == 'slic3r') {
                    _.forEach(['end_gcode', 'start_gcode'], function (key) {
                        profile[key] = profile[key].replace(/\\n/g, '\n');
                    });
                }
`

line 36555-36559
`                if (self.slicingViewModel.slicer() == 'slic3r') {
                    _.forEach(['profile.end_gcode', 'profile.start_gcode'], function (key) {
                        result$$1[key] = result$$1[key].replace(/\n/g, '\\n');
                    });
                }`

Hope this helps!

mikedmor, 2018-01-27:
I posted a fix for the problem earlier but ran into a few more issues disabling it. Instead i made the following changes to `~/.octoprint/generated/webassets/packed_plugins.js` and it seems to have solved my issue. Hopefully this help get it patched in the next version or anyone else experiencing this problem.

Change at line 36457:
`if (self.slicingViewModel.slicer() == 'slic3r') {
    _.forEach(['end_gcode', 'start_gcode'], function (key) {
        if(key in profile){
            profile[key] = profile[key].replace(/\\n/g, '\n');
         }
    });
}
`

Change at line 36555 (36557 after changing the above lines):
`if (self.slicingViewModel.slicer() == 'slic3r') {
    _.forEach(['profile.end_gcode', 'profile.start_gcode'], function (key) {
        if(key in result$$1){
            result$$1[key] = result$$1[key].replace(/\n/g, '\\n');
        }
    });
}
`

Although i want to also want to mention that this does not completely fix the problem if you have an error in your slic3r ini file, which ended up also being a problem for me. Octoprint/Octoprint-Slicer will not warn about this and the above 'fix' may cause other issues. However this did seem to get it to complete the slicing and allow me to print. Hope this helps patch it.

kennethjiang, 2018-01-27:
Awesome thank you so much @mikedmor for finding out the root cause and proposed a fix. Can you submit a PR for your fix? I'll be super busy for the following week but will try to merge your PR whenever I get a chance. Thanks again!