kennethjiang/OctoPrint-Slicer #145: import stl automatically combines with last imported stl

qqazxswedc, 2017-11-15:
while printing, when i try to import an stl(ie. clicking "upload" button) the problem is that when it opens up in the slicer, the stl was not imported on its own, but instead 'combines' with the last previously imported stl, meaning i have to select the old stl and delete it. not only that, but the really annoying part is that when i go to slice the stl, all the textboxes for file to write gcode to, etc, are filled in with the files for the last previous sliced stl, and i have to manually change them.

kennethjiang, 2017-11-15:
Hi @qqazxswedc thank you for submitting this issue.

What you described was my intentional design to make it easy for users to slice multiple STL objects at a time. You can press the button (high-lighted in the attached screenshot) when/if you want to reset it to a clean build plate.

If you have an idea for a better user experience that works well for slicing 1 object as well as slicing multiple objects, please let me know!

[image]

qqazxswedc, 2017-11-15:
okay. i see. thanks. just like desktop cura

closed 2017-11-15 by qqazxswedc