kennethjiang/OctoPrint-Slicer #147: ability to change brim line count

qqazxswedc, 2017-11-29:
the slicer is lacking the ability to change many basic settings, such as the # of lines in the brim. you can only turn brim on/off, not edit the brim settings. also, there are many settings that are changable in cura desktop edition, but not on octoprint. I would love to see more settings implemented into the plugin. thanks!

kennethjiang, 2017-12-01:
Hi @qqazxswedc - you should be able to set any tunable settings in slicing profiles. I have tried to keep Slicer plugin simple by providing the UI only for the settings that need to be changed *regularly*.

I haven't had the need to change # of lines in brim often. So I use slicing profiles for this setting and it works for me. But I understand your situation may be different. If so, can you help me understand why you need to change this setting (# of lines in brim) regularly?

qqazxswedc, 2017-12-02:
I use a printrbot simple metal, which does not have a heated bed. I often use brims, and depending on the size of the model, i change the brim amount to prevent peeling while saving filament.

qqazxswedc, 2017-12-02:
I also dont want to use up sd card space with many different profiles on my pi

qqazxswedc, 2017-12-02:
I also very often need to change the overhang angle for support. It would be great if you could add in the ability to change all the settings that are in cura, and maybe add an option in the plugin settings to turn off extra settings.

John-Mc, 2017-12-02:
@qqazxswedc - you do know that the newer versions of Cura will let you send a file directly to OctoPrint, don't you?  You can change whatever you want running Cura on your computer, slice and send the results directly to OctoPrint. If you want the **full** functionality, that's probably the best way to get it. 

I do the desktop Cura thing when I have something complex, but 95+% of the time, I don't need it. I run via the OctoPrint web interface with the @kennethjiang slicer plug-in most of the time. I also like the simplified interface for use with the elementary school where I volunteer. Presenting them with too many options when they are getting started just causes problems.

kennethjiang, 2017-12-03:
@qqazxswedc it sounds like you have a good reason to change brim line count often. I will consider adding it if I can figure out a way to prevent over-complicating the UI. 

When it comes to adding all options, I echo @John-Mc's sentiment here: the goal of Slicer plugin is to be simple and easy for 95% of the use cases out there. There are alternatives if you want exhaust solutions, such as, like @John-Mc mentioned, sending g-code directly from newer version of Cura.

Hope this makes sense.

kennethjiang, 2017-12-06:
Hi @qqazxswedc I just released version 1.3.4 that includes brim line count in settings. Please upgrade when the new version is available to you.

closed 2017-12-06 by kennethjiang