kennethjiang/OctoPrint-Slicer #152: Option to disable external tracker Mixpanel

ismangil, 2018-01-09:
When running OctoPrint with Slicer plugin enabled there is an external tracker Mixpanel enabled. How do I disable this?

If I disable Slicer plugin the tracker disappeared.

What I want is to enable Slicer without the tracker.

kennethjiang, 2018-01-09:
Thank you for submitting an issue for the Slicer plugin.

Since this is an open source software, you can just fork the source code and remove Mixpanel. It should be pretty easy.

Mixpanel has been very helpful for me to understand the usability of the Slicer plugin. For example, it helped me figure out some unintuitive designs that got users stuck.

I think calling it a 'tracker' is a misnomer. Mixpanel can't really track users. Beside telling me what buttons a user clicks, it only provides the info about user's browser (which sometimes is helpful in understanding why some users get stuck), and the city the user is in (which is often wrong anyway).

What's your specific concern about Mixpanel?

ismangil, 2018-01-09:
Ok thanks for the response.  The issue is if the internet is not connected the plugin have problem loading. But there is a workaround to disable the plugin while Internet is not connected. And as you mentioned there's always the option of forking a custom version of this slicer plugin.

closed 2018-01-09 by ismangil

kennethjiang, 2018-01-09:
Thank you for your understanding @ismangil I may be able to figure out a way to lazy load Mixpanel so that it won't stop the plugin from loading when there is no internet connection.

Thanks again for bringing this issue up!