kennethjiang/OctoPrint-Slicer #154: Add more fan options

leobel96, 2018-02-07:
Please add the possibility to disable the fan for the first "X" layers to make first layers better stuck on the bed.

kennethjiang, 2018-02-15:
Sorry for the delay @leobel96 . I have been tied up with a job transition.

Do you think it's an acceptable solution to just have the desired value set in the slicing profile? Obviously if you regularly adjust this setting, from model to model, it can be very annoying to use slicing profile for the purpose. If this is the case, can you elaborate why you need to make frequent adjust to this setting (disable fan for first X layers)?

leobel96, 2018-02-15:
You are right, it's something I'll never adjust it but I haven't cura installed on my pc so I should install the old cura, set everything and then save config and upload it. I'd like to uninstall all my slicers and only use Octoprint-slicer. In this way I can also slice and print all my projects from everywhere also from my smartphone.

John-Mc, 2018-02-15:
@leobel96 IT would be worth keeping a compatible version of Cura on your PC so you can make new slicing profiles. You never know when you might want to get into a new filament, or tweak some obscure setting that Ken's plugin does not give you access to.

leobel96, 2018-02-15:
@John-Mc Yes, it's true

kennethjiang, 2018-02-15:
If you want to be geeky about it, you can `ssh` to your `octopi` and find the `xxx.profile` file, and directly edit the settings there, as in attached screenshot.

It's a shame on everyone working on 3D printing softwares, myself included, that our users are still required to jump through all these hoops to get things done. :(

[image]

kennethjiang, 2018-02-15:
I definitely want to get to a point where users only need to go to one place to do everything, as opposed to having to keep an old software to work around problems. However, it requires a much better (and bigger) planning than this plugin warrants.

Since the original purpose of this plugin is to provide an easy and quick way for users to slice directly within OctoPrint, I hesitate want to add options that are not used often. Since you already have a workaround, I'll just close this issue for now @leobel96

closed 2018-02-15 by kennethjiang

leobel96, 2018-02-16:
@kennethjiang thank you for the suggestion!! I'll try that. Anyway I agree with everyone said. I'll edit using SSH or cura, no problem.