kennethjiang/OctoPrint-Slicer #159: Adhesion field is not working

PeggyFree, 2018-08-14:
Hello, even if I chose "None", a brim is created. I installed latest Octoprint three days ago and installed this plugin via Plugin Manager. When I chose "Advanced" -> "Adhesion" -> "None", then slice + start printing, a brim is printed. The original STL have no brim, I printed it yesterday using a Cura 3.4 STL file. 

This slicer is linked to Cura 15.04 engine. I didn't launch Cura 15.04, just let it in a folder for this Slicer. 
What's wrong?

kennethjiang, 2018-08-14:
Please post the gcode here and I'll take a look @PeggyFree 

Just in case you didn't know - there will always be a couple passes of plastic around the 1st layer. It's called "skirt". https://rigid.ink/blogs/news/3d-printed-raft-brim-skirt

PeggyFree, 2018-08-14:
I'll post the gcode asap.

PeggyFree, 2018-08-14:
Octoprint gcode - cura 15 engine

PeggyFree, 2018-08-14:
Cura 3.4 gcode

kennethjiang, 2018-08-14:
I don't see any gcode. don't think they were uploaded successfully to github. You may want to zip them before uploading.

PeggyFree, 2018-08-15:
[gcode.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/2289495/gcode.zip)
Here is the gcode.

PeggyFree, 2018-08-16:
It's a brim, isn't it?

[https://reho.st/self/7a13bf07dc5fdfada06fdef6bd738e3ff59c30d4.jpg](img)

kennethjiang, 2018-08-16:
Sorry for the delay!

Yes it was brim, not just skirt. A bit strange. Does it happen to other STLs you have?

Can you also attach the STL file, as well as the slicing profile you used?

PeggyFree, 2018-08-16:
Thank you for your answer. I'll be back with another example + STL file.

PeggyFree, 2018-08-21:
About the print speed, maybe you'll see what's wrong in the gcode, I'm not familiar with this. I changed the slicer print speed x8 (for instance 150mm/s for outer shell etc.), it didn't change anything. Then I changed the printer settings : maximum feedrate of the individual axes x10 : 90000 instead of 9000. With these settings and 50mm/s outer shell speed in slicer, I get a 2h print time instead of 20mn.

kennethjiang, 2018-08-23:
@PeggyFree You are right that even if when "adhesion" is "none", it will still generate brim in gcode.

The Slicer Plugin correctly passed the parameter `adhesion=none` to OctoPrint, which somehow passed the wrong parameter to Cura. So my best guess is that some bug in OctoPrint gets triggered by a specific setting in your Cura profile that results in the extra brim.

You can log a bug for OctoPrint at https://github.com/foosel/OctoPrint/issues/new .

A work around is to use a different `ini` file. I attached my `ini` file which doesn't suffer this problem. May you can tweak my `ini` file to fit your need?

[octave pla.ini.zip](https://github.com/kennethjiang/OctoPrint-Slicer/files/2315054/octave.pla.ini.zip)

PeggyFree, 2018-08-23:
Thank you for your analysis and help. Before testing your PLA settings i
have to solve another problem. The motors are whistling loud during print,
only when slicing with octoprint. I have to find why before anything else.