kennethjiang/OctoPrint-Slicer #167: Cura 4.4 slicer plugin

Red-M, 2020-01-27:
I've made a few tweaks to for a slicer plugin that adds Cura 4.4 with support for Cura 4.4 printer profiles.

Plugin is [here](https://github.com/Red-M/OctoPrint-PBCuraEngine).
I've been testing it and it slices files exactly as Cura 4.4 will, with all the options still available for changes (see the example printer profile folder in the git repository).

Since this is a completely up to date slicer now able to slice via octoprint, would you consider maintaining this plugin again?

jesscold, 2020-01-28:
Your repo does not have issues enabled, so expect no feedback ever or many stars.

Red-M, 2020-01-28:
Not sure why GitHub doesn't enable that if the original repository has it enabled.
I've turned that on.

kennethjiang, 2020-01-30:
I've been quite swamped these days. will take a look when I get a chance. Thank you folks!

Red-M, 2020-01-30:
I also have got a patch that adds dynamic slicing variables to your plugin @kennethjiang so that my slicer plugin can work with your UI. Its not exactly pretty but it works well.

Red-M, 2020-02-01:
I've just made [a fork](https://github.com/Red-M/OctoPrint-Slicer) that allows the new slicer to work.
The patches do make the UI for other slicers a little wonky but other than the appearance, it works really well.

The patches also allow for dynamic slicer variables as well.

kennethjiang, 2020-02-13:
I was hoping I'll be more free this week but only getting busier. Sorry folks! maybe toward the end of February

Rbn3D, 2020-06-22:
Hi!

Any updates on this? Sounds promising 👍

Red-M, 2020-06-22:
The issue I've found is with the function/formula based values.
I have an idea for solving this issue since its causes silent failures in the gcode output and affects ALL versions of curaengine, since curaengine doesn't do any of the value resolutions for formulas causing invalid slicing settings. (note; it also affects the docker container version as well).

The only work around is to resolve every single slicing option and hand each one to curaengine in the `-s` option (iirc).

Rbn3D, 2020-06-22:
So we need our own "hierarchical value resolver" that mimics cura behavior on top of curaprofile parser itself, and then all absolute values to curaengine... Why doesn't it do that by itself? 😆 

I haven't digg into this too much, but there may be another way to solve this:

Write a Cura plugin (for the desktop version) that exports full printing settings in a format we can easily parse on our side (octoprint). I know it's not ideal (support curaprofile directly would be easier for end users), but it can be a solution, and it seems fairly easier to develop.

For reference, this repo: [5axes/CuraSettingsWriter](https://github.com/5axes/CuraSettingsWriter), which is in active development, seems to do just that.

I discovered it reading this thread; https://community.ultimaker.com/topic/21945-output-print-settings-to-text-file/page/2/

Red-M, 2020-06-22:
This isn't hard to do, I just need to write some regexes to do a few things to substitute variables out with values and then run a python `exec`/`eval` over them. I just haven't gotten around to it and I'd want to do it properly.

I wouldn't want to do a settings export because then formula values would remain static forever and wouldn't update properly when you change base values (eg, nozzle size).

Rbn3D, 2020-06-22:
Of course built in profile support is always better (I didn't know about the complexity of this).

Then I guess your fork (https://github.com/Red-M/OctoPrint-Slicer) it's usable with basic profiles, isn't it?

I'll give it a try.

kennethjiang, 2020-06-23:
Sorry guys for not being active in this project any more. I have been super busy with https://github.com/TheSpaghettiDetective/TheSpaghettiDetective.

But if you guys can come up with a PR that is tested, I'll be happy to merge.

Red-M, 2020-06-25:
I may not get around to this but if I do, I'll be sure to offer my patches upstream :)

eyal0, 2021-01-16:
@Red-M Why didn't you just expand on the CuraLegacy Plugin?  Was it too difficult to do?

Red-M, 2021-01-18:
Cura legacy is so old that it doesn't fit into Cura's new versioning system.
Yes it runs but there is so many improvements and new features in Cura 4, that really, Cura should be offering a usable CLI because even the docker container version isn't properly filling values due to the way that Cura now doesn't inherit values or have any defaults, _EVERY_ value that the UI has must be to be provided to Cura's CLI when you ask for Cura's CLI to slice a file.

Profiles are not transferable from the CLI to the GUI (or visa versa) due to the way that the GUI does all the config work to then provide the CLI with every value in the GUI.

eyal0, 2021-01-18:
I heard that the new profile for Cuts are so different from how Cute used to work that the Cuts Legacy plugin is unusable.

So how did you get around this problem?

Red-M, 2021-01-19:
I built a system using the profile metadata for Cura to support changing values and to present them to the user in this octoprint plugin.
That all kinda stopped when I realised that the GUI does all the expressions and math plus defaults for Cura profiles.

X-Ryl669, 2021-07-05:
Do you know [jautocalc](https://github.com/c17r/jAutoCalc) ? I'm using this on [my blog](https://blog.cyril.by/en/3d-printers/finding-the-maximum-print-speed-of-your-printer) to automagically compute formula without bothering much about writing a javascript function per formula. 

If I understand the difficulty correctly here, it might be able to solve the variable replacement and evaluation for you directly, without requiring complex regex work and writing a calculator here.

Red-M, 2022-08-10:
I'm closing this, as much as I'd love to have something on device for slicing anywhere and with any end user device, this isn't feasible with my other works.

The main issue is that there is no dependency resolution in how to "resolve"/render values to something you could pass to the Cura CLI.
There is no easy way to do this and in someway ultimaker is making sure that nothing else can use the Cura CLI with another GUI or interface even though its used internally in every platform they make releases for Cura to (its done as an exec to the CLI binary with all the parsed and rendered values provided).

As much as I hate to say it, I'd frankly suggest changing your workflow and saving a copy of the Cura profile onto your octoprint instance to make sure you're using the right profile to slice your models with.
I think something should happen in this space to make Cura 5 available as a plugin but there needs to be serious work put in to do that currently (unless Cura change how to provide profiles and values to their CLI)

closed 2022-08-10 by Red-M