kennethjiang/OctoPrint-Slicer #176: no longer works on octoprint 1.6.0

lsellens, 2021-04-28:
Looks like 1.6.0 brings built in slicing which overrides this plugin. I'm not sure if a change needs to happen in this plugin or in octoprint to get it to function as expected once again. The built in slicing is just a pop up that lets you pick slicer and profile. No positioning or profile settings changes can be made.
[image]

lsellens, 2021-04-28:
I think this is the relevant breaking change
https://github.com/OctoPrint/OctoPrint/commit/befe2e7bee6e4e59cc21e9a4afde1df28949c543#diff-f3a6e413df2ff364beacbd858613e1e79823f944cc26a74a343098d79227bd79

eyal0, 2021-04-28:
The change that you linked is from 2015.  It's unlikely that this plugin has been broken for 6 years and no one noticed.

It ought to be pretty easy to bisect the issue, if there is one.

I can try to run this plugin with the latest version of OctoPrint and find which change broken this plugin, if there is one.

lsellens, 2021-04-28:
Yeah forget about that change I linked. I apparently didn't get enough sleep last night.
I rolled my octoprint back to 1.5.3 everything works as expected again. Just update to 1.6.0 and you'll see what I'm referring to.

kg333, 2021-04-28:
Same results here when updating to 1.6.0.

mmirate, 2021-04-29:
The minified JS in this repo includes a copy of the babel polyfill. So does OctoPrint. As you may know, the babel polyfill may be instantiated only once per page-load. Hence, this plugin's JS throws an error and gives up.

mmirate, 2021-04-29:
And, __plugin_load__ doesn't exist anymore in 1.6.0.

eyal0, 2021-04-29:
> As you may know

Actually, I know very little about this stuff!  You'd be surprised.

Do you know how to fix this?  Surely this is not the only plugin with such a problem!

mmirate, 2021-04-29:
When compiling the JS, don't include babel, just rely on OctoPrint's instance of it.

I'd send in a PR myself, but I can't find the unminified sourcecode.

rudy734, 2021-05-02:
Hello
Will there be an update for Octo 1.6 ??

eyal0, 2021-05-02:
@foosel Are you familiar with this issue? It seems that the babel polyfill in this plugin conflicts with the polyfill that is now a part of OctoPrint.

Do other plugins have this problem? How are they solving this problem?

foosel, 2021-05-03:
Not familiar, no. I also didn't get any reports about this during the RC phase, or after for that matter.

Moraxan, 2021-05-05:
Hi!

I'm using this in my work as a teacher. My students use Chromebooks, and the slicing options on the platform are limitied. I've invested quite a lot of time to get octoprint and the slicer plugin to work and it did up until this version. From what I've read in this thread is that the slicer and another plugin needs to be updated in order to work with Octoprint 1.6. One solution for me is to downgrade to 15.x, but that doesn't give any sustainbility to the solution that worked until recently. I have 0 knowledge on plugin development, but I am willing to invest time in this for it to work. My questions are the following:

What needs to be done? Preferrably in order from easiest to hardest tasks.

Pointers on how to get those things done would be appreciated.
I know some programming and debugging but is far from fluent in any language. I read code better than I write.

How do I get fail logs from Octoprint? I've looked in the settings, but haven't managed to find them. I guess that's the first place to start. To see why addons fail and then adress those problems in order.

eyal0, 2021-05-05:
The javascript ecosystem is the worst part of javascript.  What a nightmare.

I've been trying to figure out how to make a version that will work for 1.6 and <1.6.  I can do either/or very easily but making one that works for both is elusive.

kg333, 2021-05-05:
> I've been trying to figure out how to make a version that will work for 1.6 and <1.6. I can do either/or very easily but making one that works for both is elusive.

Requiring a minimum of Octoprint 1.6 going forward seems reasonable.  Using the latest plugin while unable to update Octoprint would be unusual.

eyal0, 2021-05-05:
Or I could try to strip out the polyfill.  I don't know enough about javascript to do that.

lsellens, 2021-05-07:
eyal0's PR works great for me. I'm back up and running on 1.6.0. Thanks!

eyal0, 2021-05-07:
@Moraxan Looks like new version will fix your problem.  Try it out and let me know how it goes.  Thank you for being a teacher.  Let me know if I can be of further service.