kennethjiang/OctoPrint-Slicer #183: The future of this plugin (a late night rant)

jarethmt, 2022-08-27:
Hey Guys,
So I see this plugin was largely abandoned, reason being that modern version of Cura support direct print to Octoprint. This makes sense for a generally normal, single printer, single user setup, which let's be honest, represents a lot of the community. However, as the community grows and 3D printing becomes more mainstream, public demand for on-demand 3D printing services is growing, and more 3D printing businesses are popping up to fill the void. While I can see that for hobbyist use, local Cura + send to Octoprint is fine, I think there's something being largely unexplored here, namly print farms.
Consider this.... You start printing, and as more friends and family find out you print, you start getting requests for custom prints. A more come in, you decide, "eh, screw it, I'll buy another 3D printer so I can keep printing". Well, over time, one turns to two, and two turns to four. Before long, you've got a small print farm running. Then, you finally decide to hire someone on to help out. Here's where I see the potential...
Right now, more than one user environments are lacking. If you have two users, they probably both have their own workstations (laptops). Right now, there's absolutely no easy way to share Cura profiles between environments, short of export / import. There's no means of serving them up from a central source of truth. You have to export all of your Cura profiles and instruct the new person in how to import them. Then, any changes to profiles on a local machine for small tweaks (first layer height, temps, etc) results in a mismatch on the profiles between your machines, and the best you can do is export / import again to try to get back to sync.
One solution would be a "Cura profile server" of sorts where maybe Ultimaker allows you to store your profiles and download them to a different account. Per all of my research so far, however, this doesn't exist, and I can't see any signs of Ultimaker adding it. As a closed source company, we're ultimately at their mercy here, and I'm not sure it'll ever happen. I couldn't see an analog for PrusaSlicer either.
So hopefully, you can see the conundrum... We can't share profiles in a multi-user environment and keep them up to date right now in any easy manner. This right here is why I think this plugin still has a VERY valid purpose for a growing community of small hobbyist printers turned local machine shop owner. If this plugin existed, all slicing would be tied to the Octoprint instance, which is directly tied to the printer in question. That means all profiles are actually served up by the machine itself, and if someone makes a tweak to a specific machine to keep it running smoothly, anyone else that wants to slice for that machine later will have the most up to date settings used. This seems like an extraordinarily valuable use case.
I'd love to see if I could help out in maintaining this somehow, but unfortunately I'm woefully short on time right now. I'm very familiar with web apps and github, however, and could definitely maintain this plugin IFF there was actually a community here that would value this use case, and would be able to contribute pull requests to further development. I can definitely create a list of "things that need done to get things updated", and help get things on track, and help to manage features as they come in, but I really don't have a lot of time to devote to dev.
Really, the point of this issue was to raise a thought as to the long term outlook for this project, and see if anyone else is starting to share any similar sentiments. I figure if anyone else sees this issue and is possibly willing to devote some time to help bring this plugin back to life, it's worth trying to rally them together.
So, anyone else that could benefit from this use case, or is this just me?

toddmo, 2022-08-28:
What I don't understand about this plugin being considered obsolete is that this plugin allows you to slice right from inside Octoprint. Cura doesn't. 

With Cura, you have to slice it up outside Octoprint, then send it to Octoprint. So what? The slicing files and jobs aren't integrated into Octoprint. It's another app; it's not integrated like this plugin.

The value of this plugin is never having to leave Octoprint. So how exactly is slicing inside Octoprint obsolete? Explain it to me like I'm 5.

Blackbox2018, 2022-10-20:
Why  is it obsolete? with this plugin, I didn't need the pc.

jbishop129, 2023-03-19:
Just another +1 for “why is this obsolete?”. My use case can’t be too obscure: I often need to print something simple I pull down from Thingiverse, and I used to be able to do this end-to-end from my iPad while drinking coffee. Like many other veteran 3D printer users, I have extremely well dialed-in Cura profiles, where failed prints are essentially a thing of the past. Being able to just grab a .STL, drop it in OctoPrint, slice and dice and print was a dream. Now, it is a dream of the past. :( 

That said, I’m not a developer and cannot contribute to reviving this, so I’m just here to armchair quarterback this request, in the hopes someone picks it back up.

jarethmt, 2023-03-20:
So I'll weigh in here to add to my original comment: I did find that it's possible to indeed share profiles between Cura instances by signing into your Ultimaker account on all devices that use the profiles. That said, you still have to save your settings every time you make a mod, and everyone needs to be trained to download the most recent backup from the server every single time they open Cura. Would be MUCH easier with a server based Cura though, where it's served up in the browser and there's only a single set of profiles shared for everyone.

john-clark, 2023-12-19:
Over a year later and still wishing someone with the knowledge and experience to pickup and maintain this plugin.

John-Mc, 2023-12-19:
I do miss it. It was a great set up for working with the elementary school where I was volunteering.

jbishop129, 2024-01-05:
FWIW, AstroPrint does give you the ability to drag-and-drop an .STL, slice it and print it. While it is for my case feature-parity, it has some drawbacks: Additional software in the toolchain, and a cloud-based service to do the slicing. Not show-stoppers for most, but I understand some cant or wont use cloud services.

jbishop129, 2024-01-05:
I’m traveling and not in a position to look into this, but there is a fork of this plugin with some more recent activity. Anyone looked into this already?
https://github.com/Garr-R/OctoPrint-InternalSlicer

Garr-Garr, 2024-03-19:
Hello! As @jbishop129 previously mentioned I've been working on updating this plugin for awhile now and it's finally in a very usable state!

Outside of my version being functional, the main difference between this plugin and mine is that I'm using a new version of PruasSlicer (v2.6.1) in the backend instead of Cura. So anyone's Slic3r, PrusaSlicer, or SuperSlicer profiles should be compatible! I can happily report that PrusaSlicer has been running nicely on the Raspberry Pi with the same slicer GUI everyone knows and loves.

Please let me know how you like it if you decide to give it a try :)

https://github.com/Garr-Garr/OctoPrint-InternalSlicer

jbishop129, 2024-07-07:
Thanks for taking this on, such a great feature to be able to slice straight from OctoPrint! I’m getting empty gcode output, as is the original issue reporter in below issue; I believe we are both using Ender 3 printers, FWIWGCODE output is empty · Issue #7 · Garr-Garr/OctoPrint-InternalSlicergithub.com

https://github.com/Garr-Garr/OctoPrint-InternalSlicer